andrewdavidmackenzie/pigg #2: Chose the UI framework

andrewdavidmackenzie, 2024-05-10:
My initial idea was for a native UI, and I want to work more with Iced, and so that is my default choice.

Ideas were shared on having the GUI remote from a Pi device with GPIO, including potential use of Pi Pico.

That could be done with a remote native UI connecting over network/usb, or with a web UI connecting over http and a web server running on the device.

Not sure if a web server on a Pi Pico is feasible, if not, that would rule out that option.

Iced doesn't currently support web ui, so we'd either need to chose a different framework that does support web, or have a native client and a web client, and maybe some other protocol (not just http web server) between UI and device...

sundaram123krishnan, 2024-05-10:
not sure whether pico can handle the load of a server, i have setup node servers in pi zero

andrewdavidmackenzie, 2024-05-13:
Let's go with iced, and then later we can see what lightweight networking we could do between UI and "agent" that would be possible to implement the agent part on a pi pico.

GPIOConfig and GPIOState will need to be independant of rrpal, not just for UI development, but in order to have them on pi pico also.

closed 2024-05-13 by andrewdavidmackenzie

sundaram123krishnan, 2024-05-13:
done