andrewdavidmackenzie/pigg #3: Investigate existing "Remote GPIO" projects

andrewdavidmackenzie, 2024-05-10:
I believe that remote GPIO on RPi already is a thing, so investigate what is out there and if could be used, or learned from.

sundaram123krishnan, 2024-05-10:
there are lot of python remote gpio libraries available

sundaram123krishnan, 2024-05-10:
https://github.com/golemparts/rppal
there is also a rust library available for it

andrewdavidmackenzie, 2024-05-10:
Yes, I'm using rppal in my old phone project, is pretty easy to use and works well. I will correct the reference in readme.
But it's not "remote" in the sense of accessing GPIO from a difference machine

sundaram123krishnan, 2024-05-10:
yaa true

andrewdavidmackenzie, 2024-05-22:
https://www.youtube.com/watch?v=b1IJfsaedSk

https://gpiozero.readthedocs.io/en/v1.3.1/api_pins.html

https://gpiozero.readthedocs.io/en/stable/remote_gpio.html

https://www.instructables.com/Raspberry-Pi-Remote-GPIO/

https://www.reddit.com/r/Python/comments/1b0cxwc/remoteio_a_remote_gpio_library_for_the_raspberry/

https://www.home-assistant.io/integrations/remote_rpi_gpio/

andrewdavidmackenzie, 2024-05-30:
https://www.instructables.com/Raspberry-Pi-Remote-GPIO/

https://www.home-assistant.io/integrations/remote_rpi_gpio/

andrewdavidmackenzie, 2024-06-17:
I tried remote gpio. It seemed to connect, but I couldn't get the simple python led flash example to change the LED....

Want to try @sundaram123krishnan and let me know if you get it working?

sundaram123krishnan, 2024-06-18:
Sure, will try

andrewdavidmackenzie, 2024-06-25:
Consider the pros and cons of using some existing method before we decide on our networking path

andrewdavidmackenzie, 2024-07-21:
Neither of us could get remoteGPIO working so far...

If we can get it working AND it seems people use it already, or prefer it for some reason, then we can consider adding it as an additional networking option.

andrewdavidmackenzie, 2024-07-28:
If having an accessible IP is acceptable to people, then accessing over SSH maybe worth looking at.
The user experience of remote dev/run using VSCode from host to RPi over SSH is excellent, and it can even install the required binaries on the target for you once SSH connection is established, and SSH offers you some level of security for who is accessing the target also.

closed 2024-08-13 by andrewdavidmackenzie