andrewdavidmackenzie/pigg #19: Add some tests and get the coverage up and then keep it reasonable

andrewdavidmackenzie, 2024-05-14:


andrewdavidmackenzie, 2024-06-12:
WDYT about working on this, at least partially for next-release @sundaram123krishnan ?
If you agree then just add the next-release label, thanks

sundaram123krishnan, 2024-06-13:
Sure

andrewdavidmackenzie, 2024-06-14:
@sundaram123krishnan I have added a few tests for modules that don't have much UI code, and got the coverage quite high is those, but they are the minority of the code overall so overall coverage is about 27%.

The rest is not easy to test and I don't see much low-hanging fruit.

So, I'll hand over to you. 
If you can find some useful things to test, to ensure that refactoring doesn't break things then go ahead and add them.
When you are done, just close this issue.

Later we might revisit, and do more, but I think this is enough at this stage.

andrewdavidmackenzie, 2024-06-17:
I'm going to consider this one done, for now.

Let's see if we can keep the % or increase it as we go.

closed 2024-06-17 by andrewdavidmackenzie