andrewdavidmackenzie/pigg #40: Add UI config options for Input/Output sub options

andrewdavidmackenzie, 2024-05-22:
Input can be: pullup, pulldown, none
Output can be: set high (true), set low(false), not set (None)

so add those as sub-options somehow on the UI, either as part of the current widget, or as other widgets that can be used once the main mode it set?

andrewdavidmackenzie, 2024-05-22:
We need to think how to handle these also.
So far I have only added for Inputs and Outputs and you can see the "(None)" in the name on the UI...
FYI @sundaram123krishnan

sundaram123krishnan, 2024-05-22:
yaa i saw that, will think how to nest dropdown if it's possible

andrewdavidmackenzie, 2024-05-28:
I think the ability to select pullup/pulldown is needed to have the basic "read an input" working correctly, otherwise we can't configure the pin correctly, and it won't read the correct value if (for example) a switch/button connecting it to ground is used with it... So, I added the release-0.1 label.

sundaram123krishnan, 2024-05-28:
Agreed

andrewdavidmackenzie, 2024-05-31:
We can simplify the Output option to avoid the None, Some(false), Some(true) sub-options, as there will be a toggler shown to set to true or false. So it can just say "Output"

sundaram123krishnan, 2024-06-01:
Yes, makes sense

andrewdavidmackenzie, 2024-06-01:
We could maybe do something visual with pull-up/down/none on the "wire" for the pin?

Might not be obvious what it does.

Can we add tooltips to the widgets to help I wonder?

PR #108 (2024-06-01): Implement Input Sub options in Ui
Fixes #40
closed 2024-06-01 by andrewdavidmackenzie