andrewdavidmackenzie/pigg #45: Review names of I2C bus pins.

andrewdavidmackenzie, 2024-05-22:
Change names of some pins according to the I2C bus definitions from the rppal docs:

I2C buses
The Raspberry Pi 3 B+ and earlier models support three hardware I2C buses, however only the I2C bus on physical pins 3 and 5 should be used to communicate with slave devices. The other two buses are used internally as an HDMI interface, and for HAT identification.

On the Raspberry Pi 4 B, 400 and 5, three additional I2C buses are available, depending on your configuration, as explained below.

I2C0 / I2C1
On the Raspberry Pi B Rev 1, physical pins 3 (SDA) and 5 (SCL) are tied to I2C0. On every other Raspberry Pi model, they’re connected to I2C1.

The I2C bus connected to these pins is disabled by default. You can enable it through sudo raspi-config, or by manually adding dtparam=i2c_arm=on to /boot/firmware/config.txt. Remember to reboot the Raspberry Pi afterwards.

SDA: BCM GPIO 2 (physical pin 3)
SCL: BCM GPIO 3 (physical pin 5)
I2C3
I2C3 can be enabled by adding dtoverlay=i2c3 to /boot/firmware/config.txt.

SDA: BCM GPIO 4 (physical pin 7)
SCL: BCM GPIO 5 (physical pin 29)
I2C4
I2C4 can be enabled by adding dtoverlay=i2c4 to /boot/firmware/config.txt.

SDA: BCM GPIO 8 (physical pin 24)
SCL: BCM GPIO 9 (physical pin 21)
I2C5
I2C5 can be enabled by adding dtoverlay=i2c5 to /boot/firmware/config.txt.

SDA: BCM GPIO 12 (physical pin 32)
SCL: BCM GPIO 13 (physical pin 33)
I2C6
I2C6 can be enabled by adding dtoverlay=i2c6 to /boot/firmware/config.txt.

SDA: BCM GPIO 22 (physical pin 15)
SCL: BCM GPIO 23 (physical pin 16)

sundaram123krishnan, 2024-05-22:
is this required for 0.1 release? @andrewdavidmackenzie

andrewdavidmackenzie, 2024-05-22:
I thought it would be good to get them "as correct as possible" for a first showing in public... and is easy.

PR #47 (2024-05-22): Review I2C bus pin names
Fixes #45
closed 2024-05-22 by andrewdavidmackenzie