andrewdavidmackenzie/pigg #46: Pin config change implies another pin config change

andrewdavidmackenzie, 2024-05-22:
We need to decide how to handle this case...

See https://github.com/andrewdavidmackenzie/pigg/issues/45

When I2C1 function is selected on either of these two pins:
SDA: BCM GPIO 2 (physical pin 3)
SCL: BCM GPIO 3 (physical pin 5)

Then the selection for the other pin should be changed too.

Either that, or we leave them separate and have an incorrect config that should be flagged in the UI, or used alerted to it when trying to apply to HW or save...

I think probably changing both at once would be best?

But open to other ideas how to handle it.

andrewdavidmackenzie, 2024-05-22:
Got any ideas on how to handle this @sundaram123krishnan ?

sundaram123krishnan, 2024-05-22:
ig incorrect config flag would be better?

andrewdavidmackenzie, 2025-01-18:
**Describe the solution you'd like**
Offer a third layout option, that lays out the pins grouped by their logical function as configured. So GPIO pins would be together (maybe inputs grouped and outputs grouped?), if I2C was used, those pins involved grouped together, UART, SPI etc etc...

**Describe alternatives you've considered**
The existing layouts

**Additional context**
For things like uart, it would also give a bit more vertical space and allow for nicer visualizations, alongside the group, instead of the space being scattered across pins.

Maybe the user can select a function for a group of pins all at once (e.g. SPI0) and then it changes them all en-masse.

We can then change the colors for pins to match the ALT function chosen, aligned with the colors in this doc
https://drive.google.com/file/d/14keVYmzA8FGI9p2b7R-4piGvmDs5MEX_/view
section 5.3