andrewdavidmackenzie/pigg #50: Detect hardware type and revision and make available via Hardware trait

andrewdavidmackenzie, 2024-05-23:
From https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-revision-codes

We can detect the exact HW on a Pi...and we can do something equivalent on Linux/PC/Mac HW... and make it available to the GUI by adding to the Hardware trait

"Raspberry Pi revision codes
Each distinct Raspberry Pi model revision has a unique revision code. You can look up a Raspberry Pi’s revision code by running:

cat /proc/cpuinfo
The last three lines show the hardware type, the revision code, and the Raspberry Pi’s unique serial number. For example:

Hardware    : BCM2835
Revision    : a02082
Serial      : 00000000765fc593
NOTE
All Raspberry Pi computers report BCM2835, even those with BCM2836, BCM2837, BCM2711, and BCM2712 processors. You should not use this string to detect the processor. Decode the revision code using the information below, or cat /sys/firmware/devicetree/base/model.
"

andrewdavidmackenzie, 2024-05-23:
When hardware is "none" (working on a different host), maybe show a drop-down to select the HW type and revision, so the correct config options are shown...

sundaram123krishnan, 2024-05-23:
agreed!

PR #62 (2024-05-25): Detect Pi Hardware
Fixes #50
closed 2024-05-25 by andrewdavidmackenzie