andrewdavidmackenzie/pigg #57: Update readme before release-0.1

andrewdavidmackenzie, 2024-05-24:


andrewdavidmackenzie, 2024-05-27:
@sundaram123krishnan please comment things you think should be added or improved in the README

sundaram123krishnan, 2024-05-27:
We can add example usage or provide some illustration @andrewdavidmackenzie

andrewdavidmackenzie, 2024-05-29:
Maybe add an "Installing" section or INSTALLING file, although initially it will either be:
- cargo install pigg
- clone repo and cargo install (add a make target to install??)

andrewdavidmackenzie, 2024-05-30:
Add a roadmap section with top-level ideas, comment that we want input and links to specific JIRAs with a roadmap lable.

Post-release-0.1
- review pin descriptions, maybe modify response to GUI depending on Pi HW detected
- code cleanup
- add tests for UI and get coverage up
- automation of release process and publishing packages
- improve visualizations of inputs
- add controls of outputs
- expand support beyond GPIO to I2c, uart etc,  with GUI support for config of pins in groups (e.g. I2C
- investigate remote GPIO over network and complete a "piglet" binary that acts as on-hw agent for GUI, or make it a build option of "piggui" (with a rename if it doesn't have GUI?)

sundaram123krishnan, 2024-05-30:
Sounds good

PR #113 (2024-06-02): Update readme for release
Fixes #57
closed 2024-06-02 by andrewdavidmackenzie