andrewdavidmackenzie/pigg #69: Respond to message to go FullScreen from Window Manager

andrewdavidmackenzie, 2024-05-27:
On Pi, the window manager adds close, iconize and full-screen buttons to window descorations.

Close Works

Iconize works

Full-screen does not work. I assume this sends a message to Iced, which we need to respond to in some way, or it's part of app settings via Iced when creating App.

Desired Behaviour
* User can click full-screen and piggui will go full screen
* User can click again ("restore") and window returns to original place and size

sundaram123krishnan, 2024-05-27:
will find a way to implement that

andrewdavidmackenzie, 2024-06-03:
Consider reviewing layouts when doing this?

Have the window it "shrink to fit" the content, and avoid a fixed size. It would change size depending on the layout selected, and get narrower and higher on BCM layout, and avoid big black spaces on all layouts

andrewdavidmackenzie, 2024-06-12:
This now seems to work on macOS @sundaram123krishnan

andrewdavidmackenzie, 2024-06-13:
I'd check Linux, and maybe it's fixed?

andrewdavidmackenzie, 2024-06-14:
It goes full-screen in macos, linus and pi.
So I think this is fixed and I will close it.
The problem is, as we are using fixed layouts, we don't take advantage of that extra space in any way, and the version text ends up floating in the middle, not pressed against the bottom.

Once we have a more complete UI, and we know all the things we want on screen and how they resize, then maybe it will be time to use more Fill/Shrink sizes for parts of the UI and revisit that.

closed 2024-06-14 by andrewdavidmackenzie