andrewdavidmackenzie/pigg #103: Load/Save niceties

andrewdavidmackenzie, 2024-06-01:
**Is your feature request related to a problem? Please describe.**
Help a user not lose work.

**Describe the solution you'd like**
If the user has made config change since starting or last loading a config, then on load warn that they are about to be lost and offer a cancel option.

If user clicks to exit and unsaved changes, warn them and on warning give option to save.

When loaded from file, show filename in UI or as part of window title (just name, not full path).

When loaded from file, prepopulate save with filename. Or allow a save, with no dialog....

If saving to a different file than loaded from, and exists, warn about overwriting a file and its contents will be lost.

**Additional context**
Post release 0.1

andrewdavidmackenzie, 2024-06-12:
WDYT about working on for next-release @sundaram123krishnan ?
If you agree, then just add the label.

sundaram123krishnan, 2024-06-13:
Sure, I feel like this functionality should be present in next release

andrewdavidmackenzie, 2024-06-13:
Additional ideas:

- Chang config extension to be ".pigg" as piglet can load and apply them too.

- set ".pigg" as file extension to filter on in file browser when loading

- offer user to change/add file extension ".pigg" when saving a file

PR #160 (2024-06-16): Implement Load/Save niceties 
Fixes #103
closed 2024-06-16 by andrewdavidmackenzie