andrewdavidmackenzie/pigg #207: Improve readme

andrewdavidmackenzie, 2024-06-25:
See https://www.reddit.com/r/raspberry_pi/s/l2fQqFaj8V

And my reply and try and improve that before next release

HankB, 2024-06-25:
Not obvious how to follow but this will do it. (Oh... "Subscribe" button.)

Thanks!

andrewdavidmackenzie, 2024-06-25:
Check PR #209 and let me know if better, and if enough or needs to be clearer?

HankB, 2024-06-25:
That looks better.  Enough? There is never enough but at some point you need to continue with other priorities.

One thing I looked for is what libraries, if any, are in use. e.g. `WiringPi`, `pigpio`, `libgpiod` or does the code directly deal with the kernel abstractions. The reason I'm interested in this is because I've recently faced this issue in one of my projects (written in C.)

Thank you for being so responsive.

andrewdavidmackenzie, 2024-06-25:
Before I had something on our use of Iced and rppal (the rust library for gpio that uses gpiomem and others), but removed it as "an implementation detail" :-(

andrewdavidmackenzie, 2024-06-25:
added back :-)

closed 2024-06-25 by andrewdavidmackenzie

HankB, 2024-06-25:
To be honest, many would consider that choice an implementation detail. Nevertheless... I see in a post on the Raspberry Pi forums <https://forums.raspberrypi.com/viewtopic.php?t=366693> 

> libgpiod is the standard library for driving GPIOs via the GPIO character device, /dev/gpiochipN, which is the standard GPIO userspace interface for Linux.
> That interface is the only one that works on all Pi variants. Other libraries that use the obsolete sysfs interface or the /dev/gpiomem interface will not work on the Pi5.

That comment from 2024-03-16 seems not accurate as you have this working on a Pi 5, no?

andrewdavidmackenzie, 2024-06-25:
No, I don't have a Pi5, and only have tested as far as a Pi4/Pi400.

I'll pass that on to the rppal crate author.

Thanks for the heads-up!

PR #209 (2024-06-25): Update docs
Fixes #207