andrewdavidmackenzie/pigg #222: Popup dialog to capture iroh-net connection parameters

andrewdavidmackenzie, 2024-07-02:
This is needed in order to complete #106 

To connect to a remote Iroh-net node we need:
- Node ID . Something like "odvvntniz4qijaq6gdnsxuhe2wlhugiwdgkb7uqfw6zxhke7zxmq"
- List of addresses with ports, like "79.154.163.213:55772", "192.168.1.77:55772"

Then a button to connect, or a button to cancel.

When the "Connect" button is pressed, we should disable the button, and the text entry fields while it runs.
we should show a spinner and a "Connecting to remote Pi" text.

andrewdavidmackenzie, 2024-07-21:
It will need to be a String, as per nodeid CL option, like: "qayqgh6n7voy7mskik6puouixvktqhumhwl7zrrixarv3jx4azdq"

We will need to add a method to set the HardwareTarget of HardwareView, so that the subscription tries to connect to it.

sundaram123krishnan, 2024-07-24:
Yes, got it

PR #260 (2024-07-28): Implement Popup dialog to capture iroh-net connection parameters
Fixes #222
closed 2024-07-28 by andrewdavidmackenzie