andrewdavidmackenzie/pigg #233: Close Pop-Menu on Button Click

sundaram123krishnan, 2024-07-13:
**Is your feature request related to a problem? Please describe.**
The pop-menu does not automatically close when it is clicked, leading to a less intuitive user experience.

**Describe the solution you'd like**
Ensure the pop-menu closes when it is clicked to provide a smoother and more intuitive interaction.

**Describe alternatives you've considered**
Currently, the pop-menu closes when it loses focus.

**Additional context**
Currently, this behavior cannot be achieved in the `iced_aw` library. This feature might require opening a pull request to address this limitation.

sundaram123krishnan, 2024-07-15:
I recently opened a pull request in the `iced_aw` repository, which has been merged. This feature may be included in the next release of `iced_aw`.

andrewdavidmackenzie, 2024-08-02:
Able to fix this in 0.4.0?

sundaram123krishnan, 2024-08-02:
If we use the `git` version of `iced_aw` instead of `crates.io` , then definitely we can have this functionality.
Or, wait for their next release :slightly_smiling_face:

closed 2025-01-21 by andrewdavidmackenzie