andrewdavidmackenzie/pigg #309: Problems In Menu Buttons Rendering On Linux

sundaram123krishnan, 2024-08-01:
In Linux, menu rendering problem is occuring.
The issue is further elaborated on https://github.com/iced-rs/iced_aw/issues/275

sundaram123krishnan, 2024-08-02:
As this issue is in the renderer. Now the change in `Cargo.toml` seems to fix it.
In Pi 5B also it works fine, 
So closing the issue

andrewdavidmackenzie, 2024-09-20:
@sundaram123krishnan is the iced issue fixed?
you closed it, but unclear if fixed or what....
so hard to know what to do with this one.

sundaram123krishnan, 2024-09-20:
The issue was related to `tiny-skia`, so to fix we need to bump `iced` to `0.13`.

andrewdavidmackenzie, 2024-09-20:
Assuming upgrade to 0.13 will indeed fix it...
Blocked by #466

andrewdavidmackenzie, 2024-10-26:
Check if iced/iced_aw update fixes this, if not then move to a later release to re-check

andrewdavidmackenzie, 2024-10-30:
With the changes after move to 0.13.1 we have lost hover higlighting (on linux) on all the menu buttons except the version one.

So, that's a problem (on linux) with menu bar / menu root highlighting, as the others are menus and the version on is just a button.

That might point to an issue in iced_aw?

@sundaram123krishnan

andrewdavidmackenzie, 2024-10-31:
[image]

andrewdavidmackenzie, 2024-10-31:
Apart from [highlighting on hover](https://github.com/andrewdavidmackenzie/pigg/issues/560) issue, the original problem is fixed

closed 2024-10-31 by andrewdavidmackenzie