andrewdavidmackenzie/pigg #373: Version the protocol between piggui and piglet

andrewdavidmackenzie, 2024-08-16:
In our first versions, the code assumes that piglet and piggui are from the same release with compatible code.

But in the future when we add new objects to the protocol between them, or change it, it will become necessary to not only know what HW piggui is talking to but what version of piglet or what objects/commands are supported.

Rather than version piglet in the protocol, I'd prefer to version (or hash!? there's an idea) the interface definition somehow...

Whether piggui would support older version of piglet/protocol would need to be decided when we introduce some backwards compatibility

andrewdavidmackenzie, 2024-10-26:
See if we can add unit tests for the two different implementations of serde of structs between piggui/piglet and porky, to ensure binary format compatibility?

andrewdavidmackenzie, 2024-11-28:
Maybe hardware description/details fetched has a name and version number for the binary sending it (e.g. porky/0.4.0) and piggui maintains a list of versions it's compatible with?

andrewdavidmackenzie, 2024-11-30:
The app name and version is now part of the initial fetch of hardware description. 
But if the format changes that whole transfer will fail, so not so useful for versioning the protocol, more informative.

andrewdavidmackenzie, 2024-12-10:
Maybe add the protocol version number as part of ALPN (Iroh), Service type (mDNS) and USB - so that we only discover compatible versions.
Optionally also discover older versions but flag that they are incompatible.

groutoutlook, 2026-01-21:
hi... may I know is there any new progress on this?