andrewdavidmackenzie/pigg #397: Add pin layout field to HardwareDescription and change layout accordingly

andrewdavidmackenzie, 2024-08-27:
RPi pins alternate 1 on left, 2 on right and so on, so 40 is bottom left
Pico physical pins could 1-20 down left side, then 21-40 UP right side, with 40 top right

Thus, for the board layout to be correct, we need to do it differently for RPi and Pico.

We should send the layout as a field in HardwareDescription and then change the layout view to do one way or the other.

andrewdavidmackenzie, 2024-08-30:
I have added this field: PinLayoutScheme in the HardwareDescription for pi_hw, fake_hw and porky in my pico-piglet branch.

@sundaram123krishnan would you like to work on the hardware layout view in this branch, changing how pins are laid out according to that field?

andrewdavidmackenzie, 2024-08-30:
[image]

andrewdavidmackenzie, 2024-08-30:
Pi layout, using the Rows scheme

[image]

sundaram123krishnan, 2024-08-30:
Will do @andrewdavidmackenzie

andrewdavidmackenzie, 2024-09-20:
This would be required prior to announcing/releasing Pico support is working

PR #509 (2024-10-08): Porky layout
Fixes #397
closed 2024-10-08 by andrewdavidmackenzie