andrewdavidmackenzie/pigg #440: cross build failing in GH Actions

andrewdavidmackenzie, 2024-09-13:
e.g. https://github.com/andrewdavidmackenzie/pigg/actions/runs/10850493649

I might disable cross build, so CI is green, until we can fix this...

andrewdavidmackenzie, 2024-09-13:
To work on this issue, uncomment the "cross" job in build_and_test.yml and then when working that change can be merged, so that CI should be passing before and after the change.

sundaram123krishnan, 2024-09-14:
For now we can test on armv7 architecture, will open separate for fixing cross compilation for aarch64 target @andrewdavidmackenzie

sundaram123krishnan, 2024-09-16:
I can also find a discussion related to it in `rust-skia`. 
https://github.com/rust-skia/rust-skia/discussions/959

andrewdavidmackenzie, 2024-09-16:
Need to understand what is different in local vs. GH Actions...

andrewdavidmackenzie, 2024-09-16:
Maybe we should try in local with intrument coverage?

PR #448 (2024-09-16): Cross build aarch64
Fixes #440
Fixes #392
closed 2024-09-16 by andrewdavidmackenzie