andrewdavidmackenzie/pigg #478: How to test Pi builds and HW interaction

andrewdavidmackenzie, 2024-09-25:
How could we test better the Pi builds?
- somehow build and run tests on pi arch/hw - new unit / system testing?
- detect we are running an a pi (cargo test) and run pi specific tests
- get stdout as part of tests?
- some UI checking on a display or virtual display?

How can we (automatically) check that HW interaction is working?
- detect we are running an a pi (cargo test) and run pi specific tests
- define a board config that connects an output to an input
- then write some tests that drive the output and check the input?

Have some Pi boards (with or without the external board and connections) connected to the cloud and able to accept CI jobs as custom runners?

other ideas?

sundaram123krishnan, 2024-09-25:
maybe i can try setting up pi zero for testing?

andrewdavidmackenzie, 2025-02-26:
Look at implementing a custom (local) GH Actions runner on my local Mac mini - with connected real HW - and pull a job for real HW testing when change in PRs etc?

andrewdavidmackenzie, 2025-02-27:
I have connected a pico (porky) by usb to my desktop mac mini and will always have it there.
in #884 I have implemented a series of hardware tests that detect the device via USB and then test USB and TCP connections to it.

I plan to add a piglet on raspberry pi also, and add tests for that.

I am looking into GH Actions self-hosted runners (https://docs.github.com/en/actions/hosting-your-own-runners/managing-self-hosted-runners/adding-self-hosted-runners) so that that machine can pull code in PRs and run tests on real hardware.

closed 2025-03-17 by andrewdavidmackenzie