andrewdavidmackenzie/pigg #514: A static binary would be desirable (or linking against GLIBC 2.27)

pepa65, 2024-10-09:
Can't run this on my old machine. `uname -a`: `Linux hal9000 4.15.0-1107-raspi2 #114-Ubuntu SMP PREEMPT Fri Mar 25 13:04:11 UTC 2022 armv7l armv7l armv7l GNU/Linux`

Having a static binary would solve any such issues people might have. (Or, building against GLIBC 2.27 would help people running distros of the same generation as Ubuntu 18.04.)

andrewdavidmackenzie, 2024-10-09:
Linux hal9000 4.15.0-1107-raspi2 #114-Ubuntu SMP PREEMPT Fri Mar 25 13:04:11 UTC 2022 armv7l armv7l armv7l GNU/Linux

The are not static:  
  
ldd pigg-armv7-unknown-linux-gnueabihf/piglet  
\`\`\`  
pigg-armv7-unknown-linux-gnueabihf/piglet: /lib/arm-linux-gnueabihf/libm.so.6: version \`GLIBC\_2.29' not found (required by pigg-armv7-unknown-linux-gnueabihf/piglet)

pigg-armv7-unknown-linux-gnueabihf/piglet: /lib/arm-linux-gnueabihf/libc.so.6: version \`GLIBC\_2.29' not found (required by pigg-armv7-unknown-linux-gnueabihf/piglet)

pigg-armv7-unknown-linux-gnueabihf/piglet: /lib/arm-linux-gnueabihf/libc.so.6: version \`GLIBC\_2.28' not found (required by pigg-armv7-unknown-linux-gnueabihf/piglet)

	linux-vdso.so.1 (0x7ee21000)

	libgcc\_s.so.1 => /lib/arm-linux-gnueabihf/libgcc\_s.so.1 (0x75ea1000)

	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x75e7c000)

	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x75dfb000)

	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x75de8000)

	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x75cf0000)

	/lib/ld-linux-armhf.so.3 (0x76ef3000)  
\`\`\`

Same for piggui. The lines with GLIBC not found are output when trying to run the binaries.

andrewdavidmackenzie, 2024-10-09:
Talking to colleagues more knowledgeable than me on this, it's the Linux system that enforces this dynamic loading of glibc 

When building it uses the version on the build system, which should be forwards compatible....

And I'm building on Ubuntu 24.04 I think.

One alternative for me to ensure a broader range of kernel compatibility would be to build with musl, instead of gnu - which I might look into.

A workaround will be for you to build yourself. `cargo install` will do that for you, if you have a rust tool chain....but a bit slow on your Pi... :-(

pepa65, 2024-10-10:
Building with musl is easy and works well, just add `x86_64-unknown-linux-musl` as a target: `rustup add x86_64-unknown-linux-musl`. Then `RUSTFLAGS="-Ctarget-feature=+crt-static" cargo build --release --target=x86_64-unknown-linux-musl`.

This might also work if you replace `musl` with `gnu`, but sometimes it doesn't, I don't know when it does and when it doesn't.

andrewdavidmackenzie, 2024-10-10:
I'm told there are trade-offs, but will probably try it.

This would be for x86_64 and armv7 and aarch64 cross compiles too.

pepa65, 2024-10-10:
I think the trade-off is mainly binary size, but that's offset by musl being more compact. Since you don't need to link to anything, musl is actually great, there is no real reason to go for linux-gnu that I am aware of for static binaries.

andrewdavidmackenzie, 2024-10-10:
For this project you are probably right.

I'm told some newer kernel features are not available via musl (pam =
pluggable authentication module, nss for loading information from the
system about users and such like)

andrewdavidmackenzie, 2024-11-18:
Working on this now, to include in next release.
Problem is I don't have a linux version where this fails to check.
This is a Raspberry Pi 2 with ubuntu installed, correct?

pepa65, 2024-11-18:
> Would you be able to clone repo, run "make" to build a musl binary and try it out on your Pi3B?

Not on the little Pi itself, but can I cross-compile this on my amd64?

andrewdavidmackenzie, 2024-11-18:
Well, if I get the cross-compiling working for musl, then you should be able to...
I'm struggling where to find the package that has the "arm-linux-musleabihf-gcc" linker needed to link "armv7-unknown-linux-musleabihf" target at the moment.

sundaram123krishnan, 2024-11-18:
> Would you be able to clone repo, run "make" to build a musl binary and try it out on your Pi3B?
> 
> If not, I might have a Pi3 around somewhere...
> @sundaram123krishnan what Pi do you have?
> 

I have pi zero and pi 5

andrewdavidmackenzie, 2024-11-18:
I couldn't get the "native" cross compile to work, but have been able to build it using cross (docker or podman required) ([this branch](https://github.com/andrewdavidmackenzie/pigg/tree/musl))using:
```
rustup target add armv7-unknown-linux-musleabi
cross build -target0armv7-unknown-linux-muscleabihf"
```

I have put them into a Makefile target called "build-armv7-musl"

Would you be able to clone that branch and try the resulting binaries (piggui and7or piglet) in target/armv7-unknown-linux-musleabihf/debug ?

That is in this PR: https://github.com/andrewdavidmackenzie/pigg/pull/635

If that worked, I'd be on my way - would just have left to resolve the cargo-dist release build using cross (or get native cross compile to work in CI...)

andrewdavidmackenzie, 2024-11-18:
I merged - as you can try from a clone of master branch if you want.

closed 2024-11-18 by andrewdavidmackenzie

andrewdavidmackenzie, 2024-11-19:
Did you add that target with rustup?
(`rustup target add armv7-unknown-linux-musleabihf)

I have had no problems with cross.

Unusual error there reporting some conflict with something you have installed already...

I suspect I will have a few hicups in this next release, so probably will do some pre-release attempts to iron out the kinks.
If you want to wait, I hope to have a musl binary as an artifact of those release builds and then you can download it and try it.

pepa65, 2024-11-19:
OK, I just did `rustup toolchain remove stable-x86_64-unknown-linux-gnu` and then `rustup install stable-x86_64-unknown-linux-gnu`, now I get this after `make build-armv7-musl`:
```
cross build --target=armv7-unknown-linux-musleabihf
info: downloading component 'rust-std' for 'armv7-unknown-linux-musleabihf'
info: installing component 'rust-std' for 'armv7-unknown-linux-musleabihf'
 23.1 MiB /  23.1 MiB (100 %)  19.7 MiB/s in  1s ETA:  0s
docker: permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post "http://%2Fvar%2Frun%2Fdocker.sock/v1.24/containers/create": dial unix /var/run/docker.sock: connect: permission denied.
See 'docker run --help'.
make: *** [Makefile:81: build-armv7-musl] Error 126
```

Do you run this as root?

andrewdavidmackenzie, 2024-11-19:
No, not run as root.

Is your docker daemon up and running?

On linux, I use podman which emulates docker - but without the need for a background daemon running, and it runs smoothly.

On macOS, I frequently forget to start Docker (via docker desktop) and get an error similar to yours, have to start docker so the daemon is running, then it works.

Hope that helps.

pepa65, 2024-11-19:
Docker seems to be up, although no containers are running. But I've never needed Docker to compile Rust... But maybe podman can be used as a non-root user? I think it's a permission issue.

andrewdavidmackenzie, 2024-11-19:
OK. I am trying a 0.5.0 release now, which hopefully will build the armv7 musl target correctly. 

I have it working locally (make) or in PRs (build and test workflow) - so you can use that if you can get "cross" working.

Still failing in cargo-dist release build workflow :-( 
https://github.com/andrewdavidmackenzie/pigg/actions/runs/11913755115

When I fix that, you will have a binary you can just download and use...

pepa65, 2024-11-19:
Do you have package `libltdl-dev` installed?

andrewdavidmackenzie, 2024-11-19:
I had never heard of it. Checking with apt-cache policy, it says no.

pepa65, 2024-11-20:
I think you need it to get past the breakdown in your CI build. Basically, you need to install all Ubuntu packages necessary for the build.

andrewdavidmackenzie, 2024-11-20:
I got it working last night, and there is now a 0.5.0 release on GitHub
with an armv7 musl binary artifact.

But I cannot test it, could you?

pepa65, 2024-11-20:
Yes, it works! And 10MB is a more reasonable size..!

andrewdavidmackenzie, 2024-11-20:
I did some work on binary sizes, and halved them or more (from memory).

Great!

If you have Pi Pico W, this release also adds support for them.

PR #635 (2024-11-18): Cross build for musl
Fixes #514