andrewdavidmackenzie/pigg #653: Arch Package

sundaram123krishnan, 2024-11-20:


sundaram123krishnan, 2024-11-24:
So being an arch user myself i thought of creating an arch package for our binary `piggui` and `piglet`
So, i have successfully created a one
you may have a look and i have tested the package in my system
https://aur.archlinux.org/packages/pigg-x86_64-unknown-linux-gnu
To Download:
```sh
yay -S pigg-x86_64-unknown-linux-gnu
```

sundaram123krishnan, 2024-11-24:
use then `piggui` in the terminal to execute the binary

andrewdavidmackenzie, 2024-11-24:
Are there any files that would enable "desktop integration", so that you could see it in applications, or taskbar or some such thing for Arch?
- also, is it possible to specify a logo for each binary?

andrewdavidmackenzie, 2024-11-28:
Since this is independant of the release and this milestone - removing from the milestone, and it can be done anytime