andrewdavidmackenzie/pigg #663: Waveforms in the first row of pins are clipped when running piggui on WIndows

andrewdavidmackenzie, 2024-11-25:
piggui for Windows, with porky on Pico W: top part of display not visible in either Physical or Logical layouts.

The timeline/histogram display for GPI0 does not display the On/High/True value of the Pin.

[image]

[image]

DeeJay, 2024-11-26:
Some more screenshots of failing timelines.

[image]

[image]

[image]

But this one looks correct.
[image]

All produced using piggui on Windows 10, accessing an RPi Pico W running 'porky'.

andrewdavidmackenzie, 2024-11-26:
More screenshots with problems

[image]

[image]

[image]

But sometimes is OK

[image]

Thanks @DeeJay

andrewdavidmackenzie, 2024-12-17:
I have now been able to build on run on Windows 11 inside Parallels Desktop on my Mac.

In the @bcp@ layout (connected to internal fake HW) I cannot get clipping, at top nor bottom.

[image]

[image]

It looks like @DeeJay you connected to a Pico W in the screenshots?

I haven't been able to do that yet from Windows, but will try.

I see you are on Windows 10 and I am on Windows 11... I wonder if that is the difference?

@DeeJay can you please check
- also happens when you just use the internal "fake" hardware option, any way to reproducde on that?
- do you have any access to a Windows 11 machine, to check if also happens there for you
- Any other suggestions how I could reproduce it, thanks...

andrewdavidmackenzie, 2024-12-17:
(I am use HEAD of master, so it is possible there is a difference there, I'll try 0.5.0 tag also...)

andrewdavidmackenzie, 2024-12-17:
I can see that there is a change in height of the row, when configuring as an input or output.
The waveform must be slightly taller than the space in the row for other things, and that could cause a problem.

andrewdavidmackenzie, 2024-12-17:
I have made changes in the PR https://github.com/andrewdavidmackenzie/pigg/pull/748
which I think are a good idea anyway, and *may* help here. 
But since I cannot reproduce, it's hard to know.

@DeeJay are you able to build and run HEAD of master (when I merge the PR) to verify if fixes it, or will you have to wait for a release?

andrewdavidmackenzie, 2024-12-17:
"Smart commit message" was too smart - re-opened pending your confirmation...

andrewdavidmackenzie, 2024-12-18:
@DeeJay sorry for bugging you, just tracking this issue to see if those changes fixed it or not.
Let me know if you are able to check with master. If not we'll do the release later and maybe able to confirm then...

DeeJay, 2024-12-18:
Sorry to seem unhelpful after flagging a possible issue, but my comment in #656 3 weeks ago still applies. "For the time being the conclusion is that I will not be able to test a fix for Windows if it requires building from source."

Other than Raspberry Pi boards running RasPiOS I only have access to Win10 machines: they are underpowered and short of memory and filestore - at this stage in their lifecycle I am reluctant to risk their stability by installing a development toolchain that I would not understand.

I will undertake to look at behaviour with the fake h/w stub using the release of v0.5.0 under Win10, but it may not be for a day or so.

andrewdavidmackenzie, 2024-12-18:
OK, no worries. Sorry I forgot you said that.
I'll take a punt, assume it's fixed, and after I release 0.6.0 - if then you could test a windows binary, that would be great, and I can re-open the issue if needed.
thanks.

closed 2024-12-18 by andrewdavidmackenzie