andrewdavidmackenzie/pigg #818: Enabling codegen-units = 1 for reducing the binary size a bit more

zamazan4ik, 2025-01-22:
Hi!

I found that in https://github.com/andrewdavidmackenzie/pigg/issues/551 you performed a great job with reducing the binary size with enabling special compiler optimizations. I found that the `codegen-units = 1` option is missing. This option can shrink the binary size even further (as a nice addition - compiler can perform more optimizations).

I have made quick tests (AMD Ryzen 5900x, Fedora 41, Rust 1.84, the latest version of the project at the moment) - here are the results for the `cargo build --release` command:

`piggui`:

* Release (current default): 18530680 byte
* Release + `codegen-units = 1`: 1746624 byte

`piglet`:

* Release (current default): 10943136 byte
* Release + `codegen-units = 1`: 10437392 byte

Clean build time also reduced a bit: from 2m 18s for both binaries to 2m 13s. Not a big difference but a nice addition anyway.

Thank you.

andrewdavidmackenzie, 2025-01-22:
Sounds great. I'll merge soon.

Thanks for your contribution.

andrewdavidmackenzie, 2025-01-22:
Timings of a clean build, on my mac M2

before:
* `cargo build --release  500.47s user 31.22s system 443% cpu 1:59.98 total`
* `size target/release/piggui
__TEXT  __DATA  __OBJC  others  dec     hex
7929856 32768   0       4297687040      4305649664      100a30000`

after:
* `cargo build --release  431.64s user 27.42s system 377% cpu 2:01.72 total`
* `size target/release/piggui
__TEXT  __DATA  __OBJC  others  dec     hex
7651328 32768   0       4297441280      4305125376      1009b0000´

closed 2025-01-22 by andrewdavidmackenzie